rehius/usk #9: Firmware check

T501501, 2023-11-25:
Just checking if this firmware will work on a switch with his 17.0.0 as I have heard of some issues.
Many thanks

Quas7, 2023-11-25:
I updated a V2 just yesterday to 17.0.0 with all-in-one updater and daybreak for the emunand followed by official update for sysnand.
No issues with all the latest tools installed beforehand.
I think, there might be some cases where a retraining of the picofly might be required - reflashing with 2.75 triggers this quick and easy.

T501501, 2023-11-25:
Thanks a lot for you quick response it is most appreciated

Quas7, 2023-11-27:
The OLED did update as well with no issues.

T501501, 2023-11-27:
Thanks for all your work in checking this out.

Ch3ck3rM0n, 2023-11-29:
Did you update the sysnand via standard updater from the Switch OFW?

Quas7, 2023-11-29:
Yes, I used the normal update procedure as for a non-modded console. For that, I pressed vol+ & vol- after power-on to bypass the picofly.

I have a fresh V2 with 17.0.0 and will install a picofly this weekend - I can give feedback on this one as well.

rehius, 2023-12-25:
It has no dependency on the HOS version. However the things like Hekate or Atmosphere must be compatible with your system version.

closed 2023-12-25 by rehius