rehius/usk #10: 16.0.1 picofly

BraindeadBrick, 2023-12-17:
Hi my system is 16.0.1 should i upgrade to 17.0.0 if i want to mod with picofly ? cause i heard that ppl have problem with pico on 17.0.0 and hats package or should i mod on 16.0.1 as is now

Quas7, 2023-12-18:
Two consoles with upgrade after mod and one console with 17.0.0 before mod.
See here... https://github.com/rehius/usk/issues/9

I would upgrade first and mod afterwards to safe time.

BraindeadBrick, 2023-12-18:
But isnt the newest upgrade 17.0.1? would it just work to mod after that? if i should upgrade after mod how would i do it on sysnand? and which one should be first sysnand or emunand im sorry for all the questions i really dont want to brick my black pokemon switch lite xd

Quas7, 2023-12-18:
Just update emunand with daybreak first and afterwards keep Vol+&Vol- pressed while power on. This disables the picofly and you can update sysnand as usual.

BraindeadBrick, 2023-12-19:
so my emunand will be on 17.0.0 and sysnand on 17.0.1 right cause i want the 17.0.0
So solder> hats pack> create emunand> update emunand> update sysnand gg

Quas7, 2023-12-19:
> so my emunand will be on 17.0.0 and sysnand on 17.0.1 right cause i want the 17.0.0
> So solder> hats pack> create emunand> update emunand> update sysnand gg

I never tried to use other revisions in emunand than sysnand. Some fuses might differ so patches have to be applied by Hetake, I guess.

BraindeadBrick, 2023-12-19:
its because there is no sigpatches for 17.0.1 so i want to make 17.0.0 on both just dont know how on sysnand i guess or maybe im thinking about this whole thing wrong idk

Quas7, 2023-12-19:
I think, patching the sysnand to 17.0.0 instead of 17.0.1 would just require going online with picofly booting sysnand and use daybreak instead of the official nintendo patching.
But it might be wise to DNS99ed in sysnand before booting it with picofly active to not risk to get banned for some reason.
https://gbatemp.net/threads/90dns-dns-server-for-blocking-all-nintendo-servers.516234/

rehius, 2023-12-25:
Major system updates must be done using the unmodified system to properly burn fuses. Use either reboot-OFW, or "Full Stock" mode.

closed 2023-12-25 by rehius