rehius/usk #11: Picofly does not glitch with rebuild nand(original prodinfo present) but boots into ofw.

hanswurst31, 2023-12-23:
Picofly does not glitch with rebuild nand(original prodinfo present) but boots into ofw.

Error code:

Picofly FW 2.75 short long short
Picofly FW 2.76 long short long short

-The installed picofly glitch to hekate with corrupt nand.
-The picofly glitch to hekate with empty emmc.
-The picofly(including same flex) glitches fine in another switch where the original nand is present.

-NOT SURE: i might be remembering wrong but the picofly glitches to hekate with rebuild nand WITHOUT original prodinfo present.

Tested with erista unpatched/patched and Mariko. Not tested with lite and OLED.

I use @sthetix tutorial. I asked him what the issue could be but he unfortunately told me, that he doesn't use picofly.

Do you guys think that this is fixable by FW update for picofly?

Quas7, 2023-12-23:
Did the picofly blink a few times slowly blue after first startup after flashing latest fimeware? If not, a connection is likely bad.

Normally, it should glitch and boot the payload if all connections at least to the CPU side are present (eMMC can even be removed). I had corrupted/unavailable eMMC (CMD resistor partly disconnect on OLED) and I was able to reach hetake as usual but eMMC was not showing up.

hanswurst31, 2023-12-23:
No it doesn't. Installation is perfect. I never have problems with picofly installation when orig nand is present. Even OLED always works.

Btw tested with hwfly also. Same result.

Quas7, 2023-12-23:
Than it seems not to be picofly related, if HWFly does not resolve it as well.

hanswurst31, 2023-12-23:
It's Nand structure related it seems. Picofly fw doesn't seems to be able to handle the nand structure of a rebuild nand.

Quas7, 2023-12-23:
if HWFly can not, why should picofly be able to do it?

hanswurst31, 2023-12-23:
Erista unpatched/patched and Mariko. Not tested with lite and OLED.

Dump Nand(dumping prodinfo and prod keys is enough) via Hekate and proceed with the following tutorial:

https://m.youtube.com/watch?v=FDAOghK2tHE

Then flash the rebuild nand via nxnandmanager/Hekate to emmc. 

A previously "bricked" console will boot into ofw after this. The picofly/hwfly just doesn't glitch to hekate anymore afterwards.

Quas7, 2023-12-23:
Nice to know, that rebuilding is already possible just via USB-C.
Could be, that during mmc_initialize() the emmc response is not fitting to the expected result.
For example as it tries to step through the emmc tables with this for-loop: 
`for (int i = 0; i < sizeof(mmc_init_table)/sizeof(mmc_init_table[0]); i += 4)`

Are really all partitions available and accessable on the rebuild nand? Tegraexplorer can browse them all?
[image]

hanswurst31, 2023-12-24:
Just tested a console with a completly dead emmc. So I rebuild a Nand without orig prod info present and it glitches fine.

Tutorial:

https://youtu.be/mH8osmtPFlA

That's what I was suspecting. Maybe the rebuild nand is written in a slightly different position where the switch can handle it but picofly cant. Just talked with sthetix (tutorial creator) and he told me that instead of restoring the rebuild nand via nxnandmanager straight to the emmc, I could restore it via restore section of hekate. Maybe this will solve the problem.

 I would guess that every partition is available otherwise the ofw wouldn't boot. Unfortunately I don't have those console here anymore. I will factor reset, update a working unpatched erista to the latest fw and rebuild a Nand. Then I can check and also compare both Nand. Maybe you are interested in the nand copies?

Quas7, 2023-12-24:
I never tried with any donor images to rebuild - always had all backups till now.l and used hetake.

I am not really deep into picofly firmware dev so the images would not help me much.

rehius, 2023-12-25:
> Picofly FW 2.75 short long short
> Picofly FW 2.76 long short long short
Either hardware issues or unsupported eMMC, try another eMMC here.  
It does not matter what is written within the eMMC, should always work  
However there is one intentional thing - the first boot after system update / eMMC BOOT rewrite is done into the OFW to properly burn fuses. But the following boots should go into hekate / preloader since the eMMC is burned with the custom loader.

closed 2024-08-30 by rehius