rehius/usk #12: LED bug on the Waveshare (copy) RP2040-Tiny v1.1

Yoti, 2024-03-10:
Hello!

I have a Waveshare (AliEx copy w/o Ws logo) RP2040-Tiny. It has "Rev1.1" mark on the bottom side. There is a problem with LED indication on it.

1. FW 2.67 can output light, but it will ends with infinite green even without connecting the board (only USB to PC)
2. FW 2.70+ has no LED output at all, but according to [Reddit](https://www.reddit.com/r/SwitchPirates/comments/191sp4m/no_led_light_on_rp2040_tiny_after_successful_mod/) it can be fixed via removing some resistor near the LED

There are some docs for you:
- [RP2040-Tiny Schematic](https://files.waveshare.com/upload/7/7a/RP2040-Tiny_Schematic.pdf)
- [V1.1 Schematic](https://files.waveshare.com/upload/7/7f/RP2040-Tiny_V1.1_SCH.pdf)
- [RP2040-Tiny Adapter V1.1 Schematic](https://files.waveshare.com/upload/3/35/RP2040-Tiny-Adapter_V1.1-SCH.pdf)
- [Working RGB LED sample](https://files.waveshare.com/upload/5/58/RP2040-Zero.zip) (tested on my v1.1)

Feel free to send some test builds!

Regards,
Yoti.

Quas7, 2024-03-11:
Please check, if the D_IN pin of the LED has a resistor connected before going to the RP2040 chip. You might have to bridge (short) this resistor.

See also https://github.com/Ansem-SoD/Picofly/issues/27#issuecomment-1752171010

martimparente, 2024-03-11:
Just tried the @Quas7 solution and it worked on 2.73 and 2.76
Thanks @Quas7!

@Yoti Just remove this resistor
[image]

Yoti, 2024-03-11:
@Quas7 @martimparente the LED is working OK, but not on the Picofly. I don't want to break anything, I made a bug report for the developer.

Quas7, 2024-06-23:
> @Quas7 @martimparente the LED is working OK, but not on the Picofly. I don't want to break anything, I made a bug report for the developer.

Software can not fix hardware issues...

rehius, 2024-08-30:
Added a special case for such boards, should work fine now

closed 2024-08-30 by rehius