rehius/usk #15: 2024 oled switches

Leerz, 2024-08-24:
had a few run-ins with some oled switches that just glitches too slow - takes a long time.

glitch pattern is

instant
instant
instant
40 seconds
60 seconds
80 seconds 
instant 
80 seconds
60 seconds

best guess is it's either glitching too slow or too fast (that it couldn't catch the last saved record)

upping the resistance doesn't change the result
changing mosfet setup doesn't change the result

above is using 2.75
if using 2.74 - it will always timeout and cycle back to writing the bct and training

my hypothesis is that these new 2024 emmc's (new samsung ones) needs new timings like how the toshiba/kioxia fix was done before on the early days.

or, another fix could be to have several records of known good glitch patters

try the first success pattern twice.
then the 2nd, then the 3rd etc before attempting to use the random array pattern.

rehius, 2024-08-30:
I might look into this issue once either I find such console or someone gives me a high-speed logic analyzer dump of the eMMC lines behavior of this specific case

Leerz, 2024-08-30:
> I might look into this issue once either I find such console or someone gives me a high-speed logic analyzer dump of the eMMC lines behavior of this specific case

or we could a similar fix you did with toshiba. i noticed when i flashed the no overclock file, it always reach the number of attempts, writes the emmc again and tries to glitch.

hexdra, 2024-12-02:
I also have this problem, I was on 2.7.3 for weeks and it has the same behaviour as 2.8.0. But 2.8.0 sometimes boots straight into stock fw with error code *== (short long long).

hexdra, 2024-12-02:
I also have this problem, I was on 2.7.3 for weeks and it has the same behaviour as 2.8.0. But 2.8.0 sometimes boots straight into HOS with error code *== (short long long).

Quas7, 2025-03-01:
maybe comparing the datasheets of the new and old Samsung MMC for the MMC signal drive capabilities is worthwhile. 
I would suspect, that the new MMC is more allergic to impendance issues of the DAT0/CLK wiring. 

So as a shot in the dark just routing wires over metal shielding or providing another close-by GND path via a parallel GND wire (soldered to GND on both ends) could change this behavior. Adding 47Ohm in Series also might help.