rehius/usk #16: RP2040 SD initialization issues

Paulwk13, 2024-09-27:
So I'm currently working a switch consolizer that relies on a custom made PCB that extends the SD card, I have several chipped boards but only one with a switch made 2040 and its unable to init the SD card. Forgive my ignorance I'm not to knowledgeable on the complexities of SDIO but it appears that rp2040 seem to be unable to initialize the sd card via the extender board. I have tested this extender with several alternatives and even my computer all working fine. Any insight you have could help me solve this problem. Thank you. 

[image]
[image]

Paulwk13, 2024-09-27:
[image]
[image]

This is the same board combo with one hwfly and one rp2040

Leerz, 2024-10-06:
> IO but it appears that rp2040 seem to be unable to initialize the sd card via the extender board. I have tested this extender with several alternatives and even my computer all working fine. Any insight you have could help me solve this problem. Thank you.
> 
> ![IMG_6380](https://private-user-images.githubusercontent.com/40131143/371398555-126140b3-f358-4584-9111-7bf078470cab.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyMjAyNzksIm5iZiI6MTcyODIxOTk3OSwicGF0aCI6Ii80MDEzMTE0My8zNzEzOTg1NTUtMTI2MTQwYjMtZjM1OC00NTg0LTkxMTEtN2JmMDc4NDcwY2FiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA2VDEzMDYxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc3YWQ0ZTMwYzUzZjdjZDM1MmY4NGFhMzAzNmY2MDU1M2I1ZjYyOGFjNjcwMGJiMjY4NTNkYzViNTM0NzMzYzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Yt0xqlx1e37bxLgAdCRFXc_pNEtANORmJQJCwYPzrts)
does it work when plugged in directly? 
i'm guessing the extender board also has a cap setup on the 3v line that could be causing interference. you can skip that extender board and use a generic line per line with no additional circuitry.

direbedlam, 2025-09-01:
I too am having issues with the consolizer mod from Set using the sd card extender, did you have any luck figuring out your issue?

chackAJMCPE, 2025-09-14:
I dont think it's a problem with the rp2040 firmware, rather with the length of the adapter cable. The firmware is there to upload the SDLoader and make the switch execute it.
Try connecting the sd card directly. (Or putting the sd card port elsewhere, more closer, without using a flex cable)