rehius/usk #17: another embended system

Gpy13, 2024-10-29:
Hello there!

I'm not really good in understanding code in C.
Have a question.
Do you think if I'll try connection to the V2 from orange pi GPIOs and run the glitch.c  it'll glitched?
As I see need only to modify addresses to pins and give them (pins) output mode.
Am I right or not?
Orange pi has ubuntu/debian and kernels from 4.4 to 6.1 as OS and RK3399 as SOC.
WiringPI lib is used to configurate gpios.

h3xcat, 2024-10-29:
The glitch relies is using a programmable IO which I don't believe is available on Orange PI. Using standard GPIO libraries will not be fast enough to interface eMMC.

Quas7, 2024-10-29:
Very polite and short answer from h3xcat.

@Gpy13, sorry to say pal but you are on the wrong track in quite a few ways and that makes me think that it is very likely that such a project would be way above your head. In my opinion, you would need a broader education and experience in hardware/software development to tackle this very very tough nut of a project (it is a 6 out of 10 in engineering-difficulty in my opinion even with pre-work on a silver tablet here). 

So, if you are not a direct decendent of geohot (a.k.a. George Hotz) I would advise to start with something less frustrating. And if you are - do something more worthwhile with this genius IQ!

As h3xcat explained already, even for a professional with experience in glitching a CPU to inject code it would be a rather unusual setup to use an RK3399-based SBC but I also saw somebody using a Pi4 for glitching - so what do I know.  ;P
But more commonly a more reproducable and predictable microcontroller with solid nanosecondish timing is better suited for this task as todays consols work with a few 100Mhz data bus frequencies. 

So in short: the magic trick is not the glitch.c alone but also the emmc.pio.
Thats why you normally see FPGAs (or CPLDs in former days) as mod chips for consoles as timing is of the essence for efficient glitching and code injection.

But if you are really dedicated to understand more about this topic I would advise you to just check youtube on "cpu glitching code injection" and maybe even better start googling and reading, i.e. study it.
E.g. a general overview I found here: https://www.nccgroup.com/sg/research-blog/an-introduction-to-fault-injection-part-13/

Cheers!

Gpy13, 2024-10-29:
Thank you, guys!
Another thanks for fast reply!

closed 2024-10-29 by Gpy13