rehius/usk #20: VREG_VOLTAGE_1_30 maybe not needed for 200Mhz overclock?

Quas7, 2024-12-05:
I noticed that with the 300->200Mhz change the vcore voltage was kept at maximum with 1.30V.

AFAIK a moderate 200Mhz overclocking does not require overvolting and it might be worthwhile to reduce this to 1.1V stock or maybe 1.15V to have some margin. 
This might add some stability for certain installations as it reduces the power consumption and its variability and could increase operational reliability for boarderline rp2040 cores. But your milage might vary.

Quas7, 2024-12-07:
I just tested 1.15V on v2.80 with an OLED and a Mariko together with an rp2040-tiny as a board autodetect test and so far no issues booting up just fine.

Quas7, 2024-12-15:
updated another Mariko with 1.15V - no issues so far.

rehius, 2025-01-30:
It depends on the rp2040 chip you have. Some batches may work fine, some not. So it's better to keep the voltage high just in case. This does not affect the power consumption much.

Quas7, 2025-03-01:
also a valid point. If I ever get a 1.3V working with a 1.15V update or get issues the other way around I will report back here but so far it seems quite a cosmetic issue.

closed 2025-03-01 by Quas7

souldbminerr, 2026-08-16:
@rehius Raspberry Pi has officially certified 1.15V so now we can safely use it.