rehius/usk #21: Not booting to OFW after 19.0 Update

Ixolaro, 2025-01-25:
Hey! I don't know if it's just my fuses that didn't burn or anything else, but i get a black screen on VOL+ and VOL-. Booting via Hekate or anything else works... So i believe I have to update the chip, right? (PicoFly FW 7.4.3)

Ixolaro, 2025-01-25:
To clairify, I was on 18.0 before update, and everything was working.

Quas7, 2025-01-25:
Sounds like you updated somehow the fuses but not the OFW itself or the other way around?
Booting into CFW via Hekate and updating there again might fix it maybe.

rehius, 2025-01-30:
Looks like the fuses haven't been burned properly during the system update (you can check the current fuse in Hekate to ensure it matches with the OFW version installed). I think the easiest way to fix that is to perform a system wipe whilst in a "Full Stock" mode (Hekate - Advanced)

The long way without a system wipe is to downgrade the system to 18.0 using daybreak, flashing proper BOOT0 & BOOT1 using hekate (since daybreak don't touch them), you can extract BOOTs from OFW files, there was a tool for that, and finally performing the system update again.

Ixolaro, 2025-04-15:
This has been fixed! For some reason, updating from Atmosphere doesn’t burn fuses or does something that prevents booting without CFW

Ixolaro, 2025-04-15:
So I just downgraded, and updated from OFW.