rehius/usk #22: Slow glitch

FUCKS3CURITY, 2025-02-20:
on latest firmware with oled switch and picofly. takes 20 seconds to glitch.

FUCKS3CURITY, 2025-02-20:
I see in the documents of the other git that explains added extra wires from a few capacitors but where do they lead up to?
 mentions SDA and SCL but no explanation as to where these wires go to afterwards

snowflame0, 2025-03-01:
I also seem to have a slower glitch on v2.80, Mariko switch. 

I rolled back to the last version I used for these, v2.75, On v2.75, after 2 re-boots, I see the Hekate screen in under 5 seconds again.

FUCKS3CURITY, 2025-03-01:
Im in the works of creating a new modchip that uses the riscv version of the rp2350 which is a faster and better chip, I've ordered the pcbs and the parts so hoping it makes a difference. may see an improvement but won't know until I get everything.

FUCKS3CURITY, 2025-03-01:
once I figure out the whole SDA and SCL situation, ill create a flexpcb that utilises this for the new modchip

Quas7, 2025-03-01:
The real infos you get normally here but it is quite a long read: https://gbatemp.net/threads/picofly-a-hwfly-switch-modchip.622701/page-377

But are you sure, you want to invest that much effort into a modchip as the Switch 2 is around the corner?

I really dont think it is the modchip hardware that is limiting the glitching effectiveness. 
I suspect it is more about the signal integrity on the 200Mhz DDR lines as wire length and series resistors as well as wire routing have a considerable impact. I mean you can easily corrupt your MMC by messing with the DAT0 wiring.

For DDR you need rising and falling signal edges on top - makes it even more unreliable. Just look into the KLM8G1GETF datasheet and you will see their requirements for the data/clock signals and we just fiddle some free flying wires with random impedance on these lines - obviously the results vary quite a lot. ;P

So, running a flex with filled-GND and close to datasheet recommended impedance is maybe the best way forward. 
But it also helps to just have the cables placed over/close to the shilding metal - should change the impendance as the current back-flow is less chaotic compared to "free air routing".

But I did never hook up a GHz scope to my slower installations to really check - I just changed the wire lenghts/routing and/or added a few 47Ohms in series with the data/clock lines and this did the trick for solid fast glitching in 100% of my installations.

Also I had rock-solid reliable results for OLEDs using using this BGA adapter https://github.com/abal1000x/emmc_adapter but you need some soldering/reballing skills/tools for that - and it takes much longer than flex. 
Tip: Not getting any flux under the MMC even while cleaning also solved issues I had with unreliable OLED installations for at least a year now.