rehius/usk #28: Direct boot to ofw- urgent

psinghvirk, 2025-07-20:
Hello I am using picofly and new firmware 2.80 everything working fine. Is there anyway to reverse the function logic. Example. Press vol+/vol- for cfw boot and by default boot( ofw with or without SD card )without pressing any key as factory default. It will be very helpful if you can please do this urgently waiting for your reply.

Quas7, 2025-07-21:
https://github.com/CTCaer/hekate/issues/98

Quas7, 2025-07-21:
After quickly reading a bit about the whole functionality of "VOL+ & VOL-" I really believe it is not based on the modchip FW but is a feature of Hekate that is somehow bypassing the recovery mode (RCM) via payload injection for OFW boot.

For a non-modded Switch this button combination enables the RCM during boot and obviously this is not handled by any modchip: 
https://www.nintendo.com/au/support/articles/how-to-update-nintendo-switch-in-recovery-mode/

psinghvirk, 2025-07-21:
Hi, thanks again for the reply — and totally understood on your point about program flow.

Just to clarify what I was referencing:

In main.c, the line:

bool force_button = detect_by_pull(1, 0, 1);

is checking the button state via GPIOs on the RP2040 side. Then a few lines later, this condition appears:

if (!force_button)
{
    // try booting cfw
    ...
}

So in effect:

No buttons held → enters the CFW boot path

Buttons held → skips it and goes to OFW

Since this logic is only used once, it looks like simply inverting the condition:

- if (!force_button)
+ if (force_button)

should cleanly reverse the boot behavior without needing changes elsewhere.
I double-checked and didn’t find any other place where this value is referenced for boot path decisions.

Let me know if I missed something or if you'd prefer this handled differently — and thanks again for taking the time to look into it.

psinghvirk, 2025-07-21:
Hi again! Really appreciate all your help and your detailed responses so far — they've been incredibly helpful in understanding the flow.

After reviewing everything and your clarification that the Pico doesn’t have access to the VOL+/VOL- buttons (and that the Switch handles boot selection depending on whether a payload is present), 

Rather than showing a "No SD card" screen when no payload is present, can we simply fall back to booting OFW directly. This keeps the behavior intuitive:

If the payload exists → CFW boots

If not → OFW boots

The Pico button (for USB boot/firmware flashing) still works as-is

If for any reason this path isn’t viable, would it be possible as a fallback to make the device shut down gracefully when no payload is detected — instead of sitting idle? Just to avoid battery drain in edge cases.

Thanks again for your time and support — let me know what you think or if you’d prefer a different approach!

Quas7, 2025-07-21:
So, in your scenario you would like remove the SD card or put in SD card without payload.bin to make sure to boot into OFW instead of VOL-buttons pressed during boot.

Unfortunately, I doubt that the pico does access the SD-card during its bootup magic. 
The pico fires the content of "payload.h" into the communication between CPU and eMMC storage and does not read anything from the SD-card as far as I can follow the code. 
In my understanding, the SD content is handled via Hekate. 
Also note that after glitching the pico is set to low-power/no-operation via a zzz() call.

But maybe it would be easier to add a little on/off or tactile switch to the reset line or power line of the modchip to deactivate it "hard"? The micro-switch could be placed next to the USB-C connector.
Or you can use a reed-switch inside the caseing completely invisible and only activate the modchip with a small magnet from the backside of the Switch on the right point.
I am not the first one coming up with reed switch ideas for the Switch but Sthetix (check his other videos) is using it for something else: https://youtu.be/y5-TPVecVsw?t=1010

But I have the feeling that you did not install the modchip yourself, right? 
If that is the case, please dont try to add that mod to it as there is a non-zero to quite high chance of damaging a perfectly working Switch. Or start training up with SMD soldering for a few weeks - everybody has to start with some motivation. ;)

Ixolaro, 2025-10-19:
Hello, where exactly are you seeing the logic